dynarithmic/twain_library #147: Question about queuing of scans in modal mode

dmccallie, 2025-03-17:
I'm using python 3 and DTWAIN to control a scanner with an ADF (test rig = Epson ET3750, a slow consumer scanner.)

For various reasons, I need to control the saving of the image (validating the image, adding a watermark, etc) so I am using the `DTWAIN_AcquireNative` call along with DTWAIN callbacks into a Python thread, where I capture the `DTWAIN_TN_QUERYPAGEDISCARD` event and then call `DTWAIN_GetCurrentAcquiredImage` to access the Windows dib, convert it to an OpenCV image, do the extra processing, and save the image as JPEG.

The callback function then returns '0' so that DTWAIN ignores further processing of the image.  This prevents DTWAIN from adding the image to the DTWAIN image acquisition array. I free the Windows dibs inside my callback, so hopefully this leaves me without memory leaks.

My question is about internal DTWAIN queuing - I am doing the above in DTWAIN's default "modal" mode, in order that my python code doesn't have to mess with the core Windows event loop.  When I simulate slow image processing by adding a temporary `sleep` command to the image processing step inside the callback, I see that the scanner/ADF keeps on scanning images. Fortunately those images appear to be enqueued inside DTWAIN.  When my sleep timer expires, the callback returns "0", and the enqueued images each subsequently trigger the callback, and everything works perfectly.

But this makes me nervous about the future when the code has to run with a higher-speed (200 impressions/minute) duplex scanner that might get far ahead of my in-callback processing.  I don't expect this to happen, but of course it might.

Is my approach sound? Or should I be doing my own enqueuing so that the callback returns immediately?

Thanks for you help,

--david

dmccallie, 2025-03-18:
Thanks.

I tried the code you suggested, but as best I can tell on the Epson, setting CAP_AUTOSCAN to 0 did not stop the scanner from continuing even though I had a sleep(30) on the callback after each image was dequeued.

Ah, what does the numPages argument to `DTWAIN_AcquireNative` have to do with this?  I have been testing with numPages = 1000, but perhaps I should be setting numPages=1 and re-issuing the call to `DTWAIN_AcquireNative' after each successful page?  That doesn't feel right with an ADF?

dynarithmic, 2025-03-18:
As to the callback, returning 0 from DTWAIN_TN_QUERYPAGEDISCARD will automatically delete the DIB.   In other words, discarding the page will literally discard the page.  So returning 0 would result in a double-deletion issue, thus causing your application to not work correctly.

The DIB array has to be there, so that you have access to the entire set of scanned images when done.   Thus the way you would enqueue this is to save the handle, and then `return 1` from DTWAIN_TN_QUERYPAGEDISCARD.  The DIB handle is only a 32/64 bit pointer, so it shouldn't impact the memory usage a great deal (the dib array and your queue having the same value).

As to your current question -- if the count can be set to 1 and it works correctly, you can loop the call to DTWAIN_Acquire...() like this:

```
while (DTWAIN_IsFeederLoaded(source))
{
    DTWAIN_AcquireNative(....) // specify 1 page here
}

```
This of course assumes that the Epson scanner can detect if paper is in the feeder.

dmccallie, 2025-03-18:
Strangely the Epson scanner keeps on scanning as long as the ADF has paper in it, even with `AquireNative` set to scan only 1 page, and with CAP_AUTOSCAN set to 0.  Perhaps the more sophisticated large Fujitsu will behave better.

On the free-the-dib question, if I return 1 from `DTWAIN_TN_QUERYPAGEDISCARD`, then the handle will stay alive, waiting for me to process it later.  When I am done, should I free the dib?  I assume so, but what about the acquisition array. Does that also need destroying?

Thanks!

dynarithmic, 2025-03-18:
I have an Avision 176U scanner, and it behaves correctly when given the page count of 1.   This seems to be a driver issue.

For the DIB. You will need to free the DIB yourself.   The only cases where DTWAIN deletes the DIB automatically are:

1) A return value of 0 from `DTWAIN_TN_QUERYPAGEDISCARD` and
2) All of the DTWAIN_AcquireFile...() functions will delete the DIB.

There is a [DTWAIN_DestroyAcquisitionArray](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_DestroyAcquisitionArray.html) function that you can call to destroy the array and the image data.  There is also an undocumented function `DTWAIN_DeleteDIB` that can delete individual DIB handles.

dmccallie, 2025-03-18:
Question on deleting the DIB - let's assume I enqueue a DIB to my image processing routine after capturing the `DTWAIN_TN_QUERYPAGEDISCARD` event. I save the DIB handle into my queue, and return 1 to ensure that DTWAIN doesn't delete the DIB out from under me.  As soon as I am finished processing the image, I use `GlobalUnlock` and `GlobalFree` to delete the DIB (to free up memory for the other incoming images.)

When I am done with this `Acquire_Native` batch, all the DIBs should be freed, but should I also then call `DestroyAcquisitionArray` to ensure that the DTWAIN_ARRAY itself is released?

Or maybe I should be using `DTWAIN_DeleteDIB` to delete the handles when I am done?

dynarithmic, 2025-03-19:
If you are deleting the DIB's yourself using GlobalUnlock and GlobalFree, you should either call `DTWAIN_DestroyAcquisitionArray` with the `FALSE` flag (so that DTWAIN does not attempt to delete handles that have already been deleted), or you could call `DTWAIN_ArrayDestroy` to destroy the array itself.