dynarithmic/twain_library #148: Failed to generate a multipage pdf

bparentpichette, 2025-04-04:
Hello,
I'm currently trying to launch a scan and output the files a single pdf file.
Whether i'm scanning 3 pages or a single one, I want a single pdf file.
My codebase is in C#.

Here a following block of code that is working, but this block of code generate the same amount of pdf files equivalent to the number of scanned pages.
For example, I scan 3 pages, so I will get 3 pdf files (test.pdf, test1.pdf, test2.pdf)
`            var scanSucceeded = TwainAPI32.DTWAIN_AcquireFile(source,
                "C:\\just_do_it\test.pdf", 
                TwainAPI32.DTWAIN_PDF,
                TwainAPI32.DTWAIN_USENATIVE + TwainAPI32.DTWAIN_USENAME,
                TwainAPI32.DTWAIN_PT_RGB,
                TwainAPI32.DTWAIN_ACQUIREALL,
                0,
                1,
                ref acquisitionStatus
            );`

I was looking using the arguments "TwainAPI32.DTWAIN_PDFMULTI", but when i'm trying it the scanner does it's job but there is no file been outputed.

Any ideas why the pdf multipage does not generate a single file at all ?
Thanks for your help

dynarithmic, 2025-04-04:
`"C:\\just_do_it\test.pdf"` -- you are missing an `\` escape character.

`"C:\\just_do_it\\test.pdf"`

Also, if you generated a log file, more than likely you would see the error that the file could not be created.

bparentpichette, 2025-04-04:
> `"C:\\just_do_it\test.pdf"` -- you are missing an `\` escape character.
> 
> `"C:\\just_do_it\\test.pdf"`
> 
> Also, if you generated a log file, more than likely you would see the error that the file could not be created for the reason being that the file name is invalid (more than likely, could not be opened). In any event, you should always generate a log file for issues like this. See `DTWAIN_SetTwainLog`.
> 
> Another test is to run DTWDEMO32U.exe and test the PDF file acquisition using that application. As all the file types being saved are multipage using that application, saving to PDF should work (the app uses DTWAIN_PDFMULTI). If DTWDEMO32U works correctly, then either the C# binding is broken (unlikely), or the application is doing something wrong.

My bad, I wrote down a typo but this one is correct on my codebase.
I'll check the FullDemo program and set a log file and will keep you updated.

Thanks for your quick response !!!

dynarithmic, 2025-04-04:
Please attach the log file.  I would like to take a look at it.

bparentpichette, 2025-04-07:
Hi dynarithmic,
Here a log file "DTWAIN_PDF.log" generated from my application using the argument "DTWAIN_PDF" when using the method DTWAIN_AcquireFile. As previously said, everything is working fine using this argument.
[DTWAIN_PDF.log](https://github.com/user-attachments/files/19625101/DTWAIN_PDF.log)

And here a log file "DTWAIN_PDFMULTI" generated with the same application where I change the argument for "DTWAIN_PDFMULTI". As previously said, the pdf is not generated and I didn't get any error while scanning. The only difference I can see is in this log file, the DTWAIN library seems to retry to generated the pdf file a second time (Line 249).
[DTWAIN_PDFMULTI.log](https://github.com/user-attachments/files/19625105/DTWAIN_PDFMULTI.log)

FYI,
- I set the log settings using the "DTWAIN_SetTwainLog" and the flags used is "DTWAIN_LOG_ALL"
- The directory where the pdf file is supposed to be dropped is the same directory as the log file
- Trying the DTWAIN_TIFFNONEMULTI isn't working. Here the log "
[DTWAIN_TIFFNONEMULTI.log](https://github.com/user-attachments/files/19625276/DTWAIN_TIFFNONEMULTI.log)

Is there any commands I can set to get more information according to the file generation process ?

dynarithmic, 2025-04-07:
1) The logs are incomplete.  The files are cut-off midway during the entire acquisition process.  For example, DTWAIN_PDF.log has a last line of:

`[2025-04-06 22:24:29] Thread [4908]             <<<===Exiting DTWAIN_ArrayCreate returning `

2) The current DTWAIN version is 5.6.2, but the version you are running is 5.4.2.  You should use the latest version.

dynarithmic, 2025-04-07:
Also, `DTWAIN_DCX` is a multipage format.  Are you able to generate a multipage file using `DTWAIN_DCX` (you will need a viewer that handles DCX/PCX files, such as IrfanView to verify the pages are created).

What is strange to me is why TIFF multipage does not work for you, as TIFF generation goes down a completely different code path, library, and basic process than PDF.  

What if you tried this with the user-interface for the scanner turned *on*, not off?

dynarithmic, 2025-04-07:
If you do get a chance to attempt to debug the source code, the saving of the PDF file happens [on this line](https://github.com/dynarithmic/twain_library_source/blob/ebe693f10af830724c1be984d4392ffb62bc1132/source/cpp/imagefun/dtwpdf/dtwainpdf.cpp#L1035).  If you put a breakpoint on that line, and the breakpoint is never executed, then that would explain why you don't see a file.

The `EndPDFCreation` gathers all of the objects, writes out the PDF dictionary, and closes the output file.  As a matter of fact, you should be able to see a partial output file being created with each temporary page being produced and appended to the output file.   Those temporary pages you see in the log being created are being appended onto the main output file.  

To see this, you could put a breakpoint [here](https://github.com/dynarithmic/twain_library_source/blob/ebe693f10af830724c1be984d4392ffb62bc1132/source/cpp/ctltr026.cpp#L106), as this is called after each page gets fed through the scanner.  You want to do this after the second page is scanned -- go to the output directory that you specified in the call to `DTWAIN_AcquireFileEx`, and you should see your output file, but only partially created.

So theoretically, the file always existed in partial form during the entire process, it is that so far you've only looked at the end results.

bparentpichette, 2025-04-08:
Hi,
so I was able reproduce the problem using the DTWDEMO32 application and get a log file.

[DTWDEMO32 .log](https://github.com/user-attachments/files/19655322/DTWDEMO32.log)

On the line 11062, there is an error and this seems to explain that the scanner isn't able to do a multi-page pdf. However, le single page pdf is correct.

I run the same app on an other developper machine  using an other scanner and he doesn't have the problem. So this should pointed at the scanner capabilities doing multi-page pdf.

I will try to dig down on the scanner side and investigate why the multi-page pdf isn't possible.
Thanks

dynarithmic, 2025-04-08:
The log indicates that the scanner driver is sending a "cancel" code back to DTWAIN.  

If you look at line 18039:

`[2025-04-08 11:19:29.046] Thread [28636]                   DTWAIN Message(HWND=000A151E, MSG=50464, Notification code=DTWAIN_TN_TWAINPAGECANCELED, LPARAM=15255640)`

The only way this could happen is if the TWAIN driver sends a TWRC_CANCEL back to the application after the page has been scanned.  

If you look at the [dtwaindefs.h header file in the DTWAIN source](https://github.com/dynarithmic/twain_library_source/blob/ebe693f10af830724c1be984d4392ffb62bc1132/source/h/dtwaindefs.h#L374), the comment states exactly what is occurring.

If you used a different scanner, you may see that the problem doesn't exist.   In particular, one that does not use WIA but instead, a true TWAIN driver.   It looks like the scanner you're using for these tests rejects any pages being processed except for the first page.  This is a driver issue, as whatever device you're using probably *does* have a document feeder, so nothing should theoretically stop the processing of each page.

The driver I see in the log is also peculiar in that the source information seems to be "generic", and with no name:

```
Decoded pDest:                             
{                                          
    Id=1141                                
    Version Number=1.0                     
    Version Language=13                    
    Version Country=1                      
    Version Info=26 June 2000              
    ProtocolMajor=1                        
    ProtocolMinor=9                        
    SupportedGroups=DG_CONTROL, DG_IMAGE   
    Manufacturer=No vendor info. provided  
    Product Family=Twain Data Source On WIA
    Product Name=WIA-No friendly name      
}                                          

```
Is this a custom driver?  What is the make and model of the scanner you are using?

dynarithmic, 2025-04-08:
[See this article on WIA drivers compared to TWAIN drivers](https://www.electronicofficesystems.com/2023/10/16/what-are-the-potential-limitations-of-wia-compared-to-twain/), especially the following paragraph:

> Despite these advantages, WIA does have some limitations compared to TWAIN. <b>WIA does not support some advanced features that are available with TWAIN, such as the ability to scan multiple pages in a single session</b> or to perform more sophisticated editing and effects.

bparentpichette, 2025-04-10:
We finally find out what was happening.
The problem isn't the Twain_library. We are using a fujitsu scanner and found out this post explaning a problem that may occurs within a specific windows update (https://www.pfu.ricoh.com/global/scanners/fi/dl/setup/psip-twain32-3300.html?MODEL=5116)

We applied the specific update and update the BIOS of the machine running the scanner and everything went back working as expected.

Thanks for you support and quick feedback.

closed 2025-04-10 by bparentpichette

dynarithmic, 2025-04-10:
Thanks for looking into this and resolving the issue.

It looks like this is a general issue with Windows 11 24H2, and not just your scanner.   Doing a web search on "WIA-No friendly name" shows several scanner models having this issue that has been resolved by applying various Windows updates.