dynarithmic/twain_library #149: How to get compressed JPEG size when using Buffered Transfer

dmccallie, 2025-04-04:
Hello,

Using python (32-bit) I've been successful in getting my scanner to run using `DTWAIN_AcquireNative` in "modal" mode, with callbacks into the python thread. In the callback, I access the Dib inside the `DTWAIN_TN_QUERYPAGEDISCARD` event callback. I then convert the dib to jpeg and save the file if it meets our criteria.  So far, so good.

Now I am trying to use the `DTWAIN_AcquireBuffered` approach, after verifying that the scanner supports JPEG compression by telling the scanner to compress with 
```python
dll_handle.DTWAIN_SetCompressionType (TwainSource, dtwain.DTWAIN_CP_JPEG, True)
```

**My question** - if compressed data is being returned using DTWAIN-mangaged buffers, where in the callback can I extract the completed JPEG, and how can I find out how large it is? I'd like to do this in the callback, so I can perform my quality assessment and skip saving images that don't qualify. I know how to extract a DIB, but what about a raw JPEG?

I've tried calling this inside the `dtwain.DTWAIN_TN_TRANSFERSTRIPDONE:` callback event
```python
dll_handle.DTWAIN_GetAcquireStripData.argtypes = [ct.c_void_p, ct.POINTER(ct.c_long), ct.POINTER(ct.c_ulong), ct.POINTER(ct.c_ulong),
                                                    ct.POINTER(ct.c_ulong), ct.POINTER(ct.c_ulong), ct.POINTER(ct.c_ulong), ct.POINTER(ct.c_ulong)]
dll_handle.DTWAIN_GetAcquireStripData(TwainSource, ct.byref(compression_type), None, None,
                                    None, None, None, ct.byref(bytesWritten))
```
but the returned values don't make sense  - `compression_type` is always 0, and `bytesWritten` is the same number for each `DTWAIN_TN_TRANSFERSTRIPDONE:` event.

Should I be managing my own buffers to get the JPEG data and total size? 

Thanks!

dynarithmic, 2025-04-04:
For JPEG, the buffer size will probably always be the same, except for maybe the last strip.  In any event, you should manage your own buffers and append the strips.

> if compressed data is being returned using DTWAIN-managed buffers

This is not the case when using compression.  DTWAIN only uses managed buffered for no compression *and* if there is no user-defined buffer being used.  So in short, DTWAIN only gets involved if the buffered transfer is "simple", i.e. DIB data, no compression, no user-defined buffer.  Thus you must use a user-defined buffer to get the JPEG data.

When using a user-defined buffer, DTWAIN actually does nothing with the buffer except give it over to the TWAIN source to fill in, and then from there, you get the data out of the buffer via the DTWAIN_TN_TRANSFERSTRIPDONE notification.   Thus you're responsible for taking the strip and appending it to a larger buffer (or append the strip to a file, etc.),

Also note that that the buffered transfer mode is the trickiest one to get working correctly.  If your effort proves successful, then ok, but just to warn you that sometimes, the effort in getting the buffered transfer to work correctly for compressed, user-defined buffers may be time consuming.

However, it is strange that the compression type comes back as 0.  It looks like the driver is resetting the TW_IMAGEMEMXFER setting on each transfer of the strip.  I will check this to see if this is the case.

dmccallie, 2025-04-04:
Thanks for the quick reply.

OK, challenge accepted, at least for now :)

I am successfully allocating global user-buffer memory and can copy the data from the "strips" as they come in to the callback, but the total size after I get the final `dtwain.DTWAIN_TN_TRANSFERDONE` event **adds up to the total size of the original uncompressed image.**  In other words, no compression appears to be happening!

My setup to start the scanning sequence is more or less like this. All return values are OK (removed from the code below)
```python
    dll_handle.DTWAIN_IsCompressionSupported(TwainSource, dtwain.DTWAIN_CP_JPEG ) #returns 1

    qualityFactor = 70  # JPEG quality factor (0-100)
    arrayLong = dll_handle.DTWAIN_ArrayCreate(dtwain.DTWAIN_ARRAYLONG, 1)
    dll_handle.DTWAIN_ArraySetAtLong( arrayLong, 0, qualityFactor)
    dll_handle.DTWAIN_SetCapValues( TwainSource, dtwain.DTWAIN_CV_ICAPJPEGQUALITY, dtwain.DTWAIN_CAPSET, arrayLong ) 

    # see what default strip sizes are
    dll_handle.DTWAIN_GetAcquireStripSizes .argtypes = [ct.c_void_p, ct.POINTER(ct.c_int), ct.POINTER(ct.c_int), ct.POINTER(ct.c_int)]
    strip_min = ct.c_int(0)
    strip_max = ct.c_int(0)
    strip_preferred = ct.c_int(0)
    dll_handle.DTWAIN_GetAcquireStripSizes (TwainSource, ct.byref(strip_min), ct.byref(strip_max), ct.byref(strip_preferred))

    # ask the scanner to send compressed JPEG images
    dll_handle.DTWAIN_SetCompressionType (TwainSource, dtwain.DTWAIN_CP_JPEG, True)

    # allocate user-managed buffer for the compressed images
    hBuffer = kernel32.GlobalAlloc(0x0042, strip_preferred.value)  # 0x0042 = GHND
    pBufferPtr = GlobalLock(hBuffer)

    dll_handle.DTWAIN_SetAcquireStripBuffer.argtypes = [ct.c_void_p, ct.c_void_p]
    dll_handle.DTWAIN_SetAcquireStripBuffer(TwainSource, hBuffer)

    acq_array = dll_handle.DTWAIN_AcquireBuffered(TwainSource,
                                        dtwain.DTWAIN_PT_DEFAULT, # pixel type
                                        1000, # num pages - has no effect in this case?
                                        0, # bShowUI
                                        1, # bCloseSource
                                        0, # pStatus
                                        )
# free memory, destroy DTWAIN, etc.
```
Can you see any reason why it's not compressing? I have attached the log file for scanning one page, if that helps.
Maybe this is just a bug in the Twain driver (Epson 3650)?

[dtwain_log.log](https://github.com/user-attachments/files/19610484/dtwain_log.log)

Thanks!

dynarithmic, 2025-04-04:
1) You should not call GlobalLock() on the buffer, as the driver will try and write to it during the scanning session.  You would want to call GlobalLock() at the time you want to process the data within the DTWAIN_TN_ notification when the strip is ready.

2) Start with showing the user interface, to ensure that this method works before attempting to scan without the user interface.

Unfortunately, DTWAIN doesn't have a high-level access to the DG_IMAGE / DAT_JPEGCOMPRESSION / MSG_yyy set of triplets that returns more information on the JPEG compressed images.  You could call [DTWAIN_CallDSMProc()](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_CallDSMProc.html)  to call the triplets yourself to access this data.  The [TWAIN specification 2.5](https://github.com/dynarithmic/twain_library/tree/master/TwainSpecification) mentions this triplet, and the arguments to use to get the data.  But that may be a big task at the python level (it is relatively simple on the C / C++).

I will probably add a high-level interface to these set of triplets, even though they will be rarely used (not in your case).

dynarithmic, 2025-04-04:
Also, check the first strip.  Does it contain anything looking like a JPEG header?  The header will have the string "JFIF" somewhere embedded in there.  If not, then the data you are getting back is not JPEG data, as mandated by the TWAIN specification.

dmccallie, 2025-04-04:
The data being returned is not JPEG.  The standard JFIF headers are not present.  Given that the total size returned is equal to the exact size of the image (H x W x 3 bytes), I'm pretty sure it's the actual raw image (though I have yet to display it to verify.)

I tried my code with GUI visible, but no change. I still get a non-JPEG, uncompressed result via the "strips." I also tested with both the 32-bit and 64-bit versions of the Twain drivers, with the same non-JPEG results.

I guess this is just a bug with my scanner. Unfortunately the only other scanner I have to test does not support JPEG compression. 

BTW, I fixed the `GlobalLock` problem, but my code was based on your comment...
>  "You may also want to call `GlobalLock` on the strip buffer before you passed it to `DTWAIN_SetAcquireStripBuffer` so that you don't have to make calls to `GlobalLock` and `GlobalUnlock` repeatedly when the strips are being transferred."

...which is in the documentation section `Acquire using strips` 

Thanks again for you support. If you have any additional ideas, let me know. (I'm not prepared to dive into DSMProc() data, at least not yet.)

dmccallie, 2025-04-05:
One more comment..   

When I use `DTWAIN_AcquireFileW()` with file control = `DTWAIN_USESOURCEMODE` and transfer mode = `DTWAIN_FF_JFIF`  the saved files are JPEG encoded as expected.

So I think the scanner can support JPEG but apparently not in "buffered" mode?

dynarithmic, 2025-04-05:
I tested both `DTWAIN_FF_JFIF` and `DTWAIN_FF_TIFF` using `DTWAIN_USESOURCEMODE` and `DTWAIN_AcquireFile`, and for the Fujitsu scanner I have, both JFIF and TIFF worked correctly, so for at least that scanner, saving to a file does work.   However I  have trouble getting the buffered JFIF to actually produce images, with varying results resulting in either a crash (Fujitsu fi series), or no JPEG data (Avision 176U, but that driver has other known bugs, since saving JFIF to a file using `DTWAIN_AcquireFile` does not work).

The main issue is that there really isn't any good help from the Twain Working Group in terms of a concrete, real example of how the buffered, compressed transfers are supposed to work.  Yes, the TWAIN specification theoretically talks about how it works, but even their open source, sample virtual TWAIN data source has commented out and left blank their buffered mode using JFIF compression.   

This is unfortunate in that there is a lot of intricate, moving parts (ICAP_JPEGQUALITY, the Huffman tables, when to exactly call DG_IMAGE / DAT_JPEGCOMPRESSION / MSG_SET and GET triplets in the grand scheme of things, etc.), and without a concrete example to work with, this becomes very difficult.

So basically, 

1) Driver authors are operating in the dark, only relying on written words as to how to implement this.  What one driver implements to get this to work may be different than what works for another driver.

2) Almost no drivers are open source where I could take a look at how to get the JPEG compression to "behave".  If I had the source code to some of these drivers, it would take me literally a few minutes to an hour to figure out what to do.

So I would say to stay away from buffered mode for JPEG, unless or until I have a handle on getting at least one to work.  If not that, come up with a "fake" buffered mode that supports buffering without the high-level code actually needing to be changed, (which may be a good idea).

dmccallie, 2025-04-05:
Thanks for looking so deeply into this.

I know opinions about performance are heavily scanner dependent but I have a theoretical question for you. I have a version of my code that works with `AcquireNative()` where I grab the completed DIB in the callback, strip the headers, copy the data to a numpy data array (removing the padding bytes) and then save as JPEG.  This works.

Do you think using the `AcquireBuffered` mode, with a user-created buffer, where the callback accumulates the uncompressed image, which I then save as JPEG would be significantly faster? Is the gain of using user-buffering and avoiding the DIB handling likely to be worth it?

Thanks!

dmccallie, 2025-04-05:
Oops, I now see that the documentation says using buffered mode with no compression would return a DIB, rather than just raw image data.  I guess that's OK, since the DIB header would have all the metadata I need.

Same question - would you expect meaningful performance differences between AcquireNative vs AcquireBuffered, for DIB image data?

dynarithmic, 2025-04-05:
Performance really depends on the actual scanner and TWAIN implementation.  Also, you should take in consideration the physical speed of the scanner, the buffering of pages (whether the scanner supports this, i.e. CAP_MAXBATCHBUFFERS), etc.

However, for safety and stability, acquiring using native mode must always work correctly for every TWAIN compliant device.  I have never encountered a device that did not implement native mode correctly.

With buffered mode, since the data is coming in as strips, hopefully you are taking care of any image alignment issues when appending to your buffer.  One mistake I see is not taking into consideration that each strip starts at offset 0 of the image row, and instead the application blindly appends the strips together.   What ends up is the "staircase" issue when the image if finally created if the strips coming in are not word-aligned. 

In native mode, you don't have this issue, as the scanner is responsible for creating the entire image page.

dmccallie, 2025-04-05:
Good point - I was assuming that incoming data was strips of JPEG, so alignment didn't matter, but with uncompressed mode, I would have to do the extra alignment work.  And extract the DIB header data from the DTWAIN call instead of the bitmap itself.

So for now, I will stick with Native mode.  It works.  We'll see if it's fast enough.

Thanks again for all the help.

closed 2025-04-05 by dmccallie