dynarithmic/twain_library #150: logging question

dmccallie, 2025-04-12:
What is the earliest that the `DTWAIN_SetTwainLogW()` routine can be called?  Does `DTWAIN_SysInitialize()` need to be called before the `SetTwainLog()` is called?

I do my acquisitions in a disposable python thread. Inside the thread I call the ` DTWAIN_SysInitialize()`, then call all the various DTWAIN functions, including `DTWAIN_AcquireNative().`  When the acquisition is done, I call  `DTWAIN_SysDestroy()` and the thread is closed.  Subsequent acquisitions create a new thread, and the process repeats. This works well so far.

I'd like to be able to keep a single log file across all the various cycles of `DTWAIN_SysInitialize()` and `DTWAIN_SysDestroy()`, but I believe the log file can only be called after `DTWAIN_SysInitialize()`?  Is there some way to setup the log file outside of the thread that does the acquisitions?

Thanks.

dynarithmic, 2025-04-14:
You can apply the [DTWAIN_LOG_FILEAPPEND](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetTwainLog.html) flag to the call to `DTWAIN_SetTwainLogW` to append to the log file.

dmccallie, 2025-04-15:
Thanks, works great.

closed 2025-04-15 by dmccallie