dynarithmic/twain_library #152: DTWAIN_AcquireFile+ DTWAIN_USESOURCEMODE with callbacks to inspect most recent image?

dmccallie, 2025-04-18:
Is it possible to use `DTWAIN_AcquireFile()` and the `DTWAIN_USESOURCEMODE ` mode (the scanner does the file saving) along with a callback function that would allow the application to inspect the image using the trigger `DTWAIN_TN_QUERYPAGEDISCARD` and `DTWAIN_GetCurrentAcquiredImage()`?

Also, in the USESOURCEMODE could returning 0 to the message `DTWAIN_TN_PAGECONTINUE` be used to stop subsequent scanning?

Thanks!

dynarithmic, 2025-04-18:
Unfortunately the TWAIN specification for `DTWAIN_USESOURCEMODE` does not allow images to be created, only files, so `DTWAIN_GetCurrentAcquiredImage` isn't utilized at all.  The only notifications you will get with `DTWAIN_USESOURCEMODE` would be `DTWAIN_TN_FILESAVEOK`, `DTWAIN_TN_FILEPAGESAVEOK` and `DTWAIN_TN_FILESAVEERROR`.   

So you probably have to devise a way to have the file saved (you can't intercept this), handle one or more of the mentioned notifications, and then have a (probably third-party) viewer to view the file to the user.  Then you would save or delete the file depending on the user's choice.

I haven't tried `DTWAIN_TN_PAGECONTINUE` and `DTWAIN_USESOURCEMODE` together, but theoretically it should work, provided that your scanner driver has multipage support (most likely `DTWAIN_FF_TIFF`).

dmccallie, 2025-04-18:
When the callback gets `DTWAIN_TN_FILESAVEOK` is there any way inside the callback to find the filename of the file just saved?

dynarithmic, 2025-04-18:
You can use the [DTWAIN_GetSaveFileName](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_GetSaveFileName.html) function.

dmccallie, 2025-04-19:
Thanks, I will give it a try.

closed 2025-04-19 by dmccallie