dynarithmic/twain_library #154: Fujitsu FI-8170 - code works OK with 64, but intermittent fails with 32 bit driver

dmccallie, 2025-04-29:
Here's a strange failure that we can't solve.  The code is a python program that works fine on the Fujitsu FI-8170 in 64-bit mode, but fails intermittently on that same scanner in 32-bit mode.  The code works fine on a different scanner (Epson 3750) in both 32-bit and 64-bit mode.  To make matters more confusing, the FI-8170 appears to work fine with your `DTWDEMO32U.exe` program.

The failure is that after zero or more successful acquisitions, the scanner simply stops sending callbacks, and the log file shows an infinite loop waiting for the scanner to do something.  I have attached a truncated log file that shows a failure on the first attempt at acquisition. Usually we get one acquisition to work before failure, but this time it failed on the first try.  

As best I can tell, the scanner stops behaving right after the callback event `DTWAIN_TN_ACQUIRESTARTED.`  The callbacks appear to stop coming, and DTWAIN loops waiting for something that never comes.  Maybe something obvious will jump out if you could take a look at the log?

We are trying to support both 32 and 64 bit machines, because our future production scanner is not yet selected, and could possibly be 32-bit only.  

The Python code itself uses `DTWAIN_AcquireNative`, in MODAL mode, with a callback into the python program, where we intercept the `DTWAIN_TN_QUERYPAGEDISCARD` event, grab the image DIB using `DTWAIN_GetCurrentAcquiredImage`, process and save the image, and then return '0' to let DTWAIN know that we have taken control of the image.  We normally run with the scanner's UI turned off, but the failures occur even if the native UI is turned on.  The acquisition runs in a separate Python thread, and all DTWAIN calls are issued from the same thread. The callback is defined as a closure inside that thread and works as expected in 64-bit mode on the Fujitsu, and in both 32 and 64 bit mode on the Epson.  So I think we are handling the callbacks properly (carefully respecting the differences between 32 and 64 bit signatures.)  

You might need more data in the log file, but we had turned off some of the details because the files were getting so large.  I can re-try with more log flags if that would be helpful.

Sorry for submitting such a vague problem! We are hoping there's something obvious that we are missing.

[initial_fail_dtwain_log.log](https://github.com/user-attachments/files/19968781/initial_fail_dtwain_log.log)

dynarithmic, 2025-04-30:
There was an [issues thread](https://github.com/dynarithmic/twain_library-java/issues/18) created for an issue with the Fujitsu 6130z scanner (which may or may not use the same base software driver as your model).   Even though the language used in that issue case is Java, it is quite possible that your scanner is exhibiting the same behavior.  The symptoms were the same -- the program works without error for C++, but for Java (and maybe for Python), something odd is occurring.  For some reason, the Fujitsu drivers are flaky when working outside of a C or C++ application.

This was corrected, or at least somewhat addressed, by changing the internal TWAIN message loop being used when acquiring an image  (change the loop to use the Windows API PeekMessage() instead of GetMessage()).  The choice of using PeekMessage() solved the issue within Java, so the ultimate solution was to create a setting in [DTWAIN32.INI](https://github.com/dynarithmic/twain_library/blob/2b7b44c02fcb8b9438050151e17589761165e6a3/text_resources/dtwain32.ini#L9) that checks the product name of the source, and if it matches the name in that setting, then PeekMessage() is used.  

Make sure that you are using dtwain32.ini with that setting on.  If you are, then as a test, comment out the setting to see if the problem happens not intermittently, but on each and every scan (probably the 64-bit version will also fail). 

As to Fujitsu drivers themselves, they have always been a pain-in-the-neck, even going back to the Windows 2000/XP days.  The drivers install themselves as services, and if anything went wrong, the OS had to be rebooted to get the driver up and running.  

I don't know if this is a bug in the Fujitsu driver, but for some reason it has always been Fujitsu scanners that had these quirks and weird behavior.

dmccallie, 2025-05-01:
So here's the results of our testing:

For the Fujitsu FI-8170:

- the 64-bit test works correctly regardless of the presence of the '[TwainLoopPeek]' setting. In other words, it always works.
- the 32-bit test works ONLY if the `[TwainLoopPeek]` setting for Fujistu/Paperstream is **commented out**. In other words, we must remove the Fujitsu/Paperstream flags to get it to work (!)

We have not selected our production scanner.  Given your reservations about Fujitsu drivers, do you have any recommendations for manufacturers with better Twain/DTWAIN support?  We need ~100 physical pages per minute, ADF, duplex scanning, plus an imprinter.

dynarithmic, 2025-05-02:
My experience with the FI-6130z is that the 64-bit driver would fail without the setting, and the 32-bit driver would work regardless of the setting.   So I guess you can either 

1) If the scanner runs consistently ok with this setting commented out, comment out the setting for your environment.  

or if you want something more flexible that you can dynamically switch at runtime, 

2) There is an undocumented DTWAIN function, `DTWAIN_EnablePeekMessageLoop(DTWAIN_SOURCE Source, BOOL bSet)` that sets the PeekMessage() processing for the source to "on", depending on the `bSet` parameter.  If `bSet` is 1, then PeekMessage is used, otherwise `GetMessage` is used.  Similarly there is a `DTWAIN_IsPeekMessageLoopEnabled(DTWAIN_SOURCE)` function to test if the 
Source is using PeekMessage() or GetMessage().

As to scanners, the 32-bit and 64-bit Avision 176U scanner has ADF, duplex, and imprinter support.   Unfortunately this is the only scanner I have in-house that has the imprinter option (the FI-6130z has an imprinter option, but I would need to get one and install it).    So I suggest sticking with the Fujitsu, and going with 1) or 2) above, since the issues I've had with the Fujitsu drivers were solely on getting them to start up, but once they started, were full-featured.

closed 2025-06-29 by dmccallie