dynarithmic/twain_library #159: Fujitsu "internal error occurred in driver" ([B2000])

dmccallie, 2025-06-10:
Hello - I have a complex question here.  I have a scanner thread which works on the first acquisition, but on some machines (e.g. Fujitsu 8170) it fails with "internal error in driver" on the second batch. It works fine on an Epson scanner. I've pasted a pseudocode outline of my code below. Sorry for all the details, but maybe it's useful. 

I am using 64-bit drivers, and use `AcquireNative()` in Modal mode, with callbacks, to extract each image inside the callback, validate the image, and stop the batch if the document fails validation. The operator has to find the failed document, fix the problem, re-insert it into the hopper stack, then start a new batch. 

It's python code. I run the scanner loop inside a persistent thread that gets spawned when the program launches.  The main program opens the DTwain DLL, sets up the GUI and a command queue, then spawns the scanner thread.

The scanner thread does one-time initialization of the scanner including: `SysInitialize()`, set modal, enable message notify, register the CB, and prompts the user for the Source using the `DTwain_SelectSource()`. It then loops waiting for commands.

When a "start batch" command is received, the scanner thread first sets various scanner parameters (pixel type, resolution, page size, duplex, ADF) and then calls `AcquireNative()` with `bShowUI=False` and `bCloseSource=False.`

Inside the callback, I catch the `DTWAIN_TN_QUERYPAGEDISCARD` event, extract the DIB, validate the image and save it. Then return 0 to let DTWAIN know to discard the image memory.  If the validation fails, I set a `stop_signal` such that on the next page, the `DTWAIN_TN_PAGECONTINUE` or `DTWAIN_TN_TRANSFERREADY` events will return 0 signaling the batch to stop.  Otherwise CB returns 1 to keep scanning.

When a "shut down" command is received, I break from the command loop and call the `DTWAIN_CloseSource()` and `DTWAIN_SysDestroy()` functions, then exit the thread and exit the program.

In general the code works as intended. The "stop" action can take a few sheets to stop the high speed scanner, but that's OK. (I will experiment with trying one page at a time scanning in the future.)  As noted above, the code works on an Epson scanner, but fails _on the second batch_ with the Fujitsu.

Perhaps I need to be closing and reopening the Source for each batch? Or calling some other re-set function? Can you see anything else wrong?

Thank you!

Pseudocode:
## main routine

```python
# load DTwain
dtwain_dll = dtwain.load_dtwaindll(path)

# create GUI thread (Tkinter...)
root = tk.Tk() ...

# create a command queue
command_queue = queue.Queue()

# spawn scanner thread
scanner_thread = threading.Thread(args = dtwain_dll, command_queue, ...)

# run main GUI loop - sends user's commands to scanner thread
root.mainloop()

# free DLL handle
dtwain_dll = None
exit()
```

### persistant scanner thread
```python
# define the python callback function
def my_callback(wParam, lParam, userData)
    
    # has stop signal been set?
    if wParam == dtwain.DTWAIN_TN_PAGECONTINUE or wParam == dtwain.DTWAIN_TN_TRANSFERREADY:
        if stop_signal.is_set():
            return 0 # stop this "acquire native"

    # an image is ready - get DIB and validate image
    if wParam == dtwain.DTWAIN_TN_QUERYPAGEDISCARD:
        handle = dll_handle.DTWAIN_GetCurrentAcquiredImage(TwainSource)
        img = extract_image_from_dib_handle(handle)
        
        status = validate_image(img)
        
        # if image is valid, save it
        if status:
            save_as_jpeg(img, path)
            
            return 0 # tell DTWAIN to DISCARD this image
        
        else:
            # image is not valid - we need to stop this acquisition
            # set a stop signal for next image
            stop_signal.set()
            return 0  # discard this image

    # all other wParams return 1 to keep processing
    return 1 # continue scanning      

#
# one time initialization when thread is launched
#
dll_handle.DTWAIN_SysInitialize()

dll_handle.DTWAIN_SetTwainLogW(log_flags, log_file)

# set modal mode and set message notifications
dll_handle.DTWAIN_SetTwainMode(dtwain.DTWAIN_MODAL)
dll_handle.DTWAIN_EnableMsgNotify(True)

# set callback (attends to 32-vs-64 bit)
dll_handle.DTWAIN_SetCallback64(my_callback, 0)

# user selects scanner via DTWAIN's popup
TwainSource = dll_handle.DTWAIN_SelectSource()
#
# loop once per "batch"
#
while True:

    command = command_queue.get() # block here wait for GUI's command

    if command == "do one scan":
        
        # set batch parameters - these never change - should these move outside loop?
        dll_handle.DTWAIN_SetResolution(TwainSource, res_val)
        dll_handle.DTWAIN_SetPixelType(TwainSource, px_type, px_depth, True)
        dll_handle.DTWAIN_EnableFeeder(TwainSource, True)
        dll_handle.DTWAIN_EnableDuplex(TwainSource, True)
        dll_handle.DTWAIN_SetPaperSize(TwainSource, paper_size, True)

        # enable autoscan, but future consider doing one page at a time 
        dll_handle.DTWAIN_EnableAutoScan(TwainSource, True)

        # start batch - blocks here till done or stop return = 0 is sent via Callback
        acq_array_handle = dll_handle.DTWAIN_AcquireNative(
            TwainSource,
            dtwain.DTWAIN_PT_DEFAULT, # Pixel type
            -1,    # Max pages (-1 for continuous)
            False, # bShowUI
            False, # <<-- bCloseSource = FALSE -->> Keep source open!
            None)  # pStatus
        
        # verify that acquisition array has no entries to free
        acq_count = dll_handle.DTWAIN_ArrayGetCount(acq_array_handle)
        if acq_count > 0:
            img_array = dll_handle.DTWAIN_GetAcquiredImageArray(acq_array_handle, 1)
            if img_array:
                count = dll_handle.DTWAIN_ArrayGetCount(img_array)
                if count > 0 print error message (this is always zero)
        
        # delete the acquisition array regardless
        dll_handle.DTWAIN_DestroyAcquisitionArray(acq_array_handle, True)

        # leave source open, loop/wait another command

    if command == "shut down":
        # break from scanning loop and let shutdown start
        break

    # close source and exit dtwain
    dll_handle.DTWAIN_CloseSource(TwainSource)
    dll_handle.DTWAIN_SysDestroy()
        
    # exits the thread which causes main program to exit as well
```

dmccallie, 2025-06-11:
Thanks for the suggestion. 
What's the best way to close the source? Is it sufficient to set bCloseSource = True in the call to `DTWAIN_AcquireNative`?

dynarithmic, 2025-06-12:
You could get a more in-depth diagnosis of where the problem occurs by producing a DTWAIN log and seeing which triplet is tripping up the Fujitsu driver.  

Also another (more advanced) mode of seeing where things break down is to get the [debug version](https://github.com/dynarithmic/twain_library_source) of DTWAIN along with the source code, attach the debugger to the running python instance, and get a look at the call stack when the error occurs to see the sequence of calls.

dynarithmic, 2025-06-12:
I found [this](https://community.spiceworks.com/t/paperstream-ip-communication-error-with-fujistsu-7160-scanner/826050) when searching for this error with Fujitsu scanners.  The suggestion by **spiceuser-pxdgz** (if available with your scanner) could be the issue.

dynarithmic, 2025-06-12:
Going back to my previous comment concerning the bCloseSource argument, it really didn't come into play until the entire scan session was ended (the source UI is closed).   So even if bCloseSource is used, it was only utilized after DTWAIN_Acquire...() is about to return to the application.   Your situation is different in that you would want to close and open the source during the processing of each batch (let me know if this is correct or not).

Admittedly the documentation states that there are many sources that do not behave correctly if not closed after acquiring, but that really hasn't been the case for most devices within, say, the last 8 or so years.  So the documentation (which is basically for DTWAIN 4.x and earlier with only updates for 5.x) should be updated with this information.  Either that, or I will put the closure of the source back into play with a later version of DTWAIN.

dmccallie, 2025-06-12:
Hi,
Sorry for the slow response - our testing requires remote site coordination and has slow turn-around.

I tried opening and closing the source around each batch, but that didn't work. The second batch just doesn't ever start scanning. It gets this far in the event stream, and then hangs:
```
Scanner: Callback: DTWAIN_TN_UIOPENING
Scanner: Callback: DTWAIN_TN_UIOPENED
Scanner Callback: DTWAIN_TN_ACQUIRESTARTED
```

I tried setting the `bCloseSource = False`, but as you expected, that didn't work.

I found some earlier testing that we had done (not using callbacks) where each "batch" was processed in a fresh thread, with the complete `DTWAIN_SysInitialize()` / `DTWAIN_SysDestroy()` cycle in each thread instantiation, and that code _had worked as expected_.

So my next test will be to move the initialize/destroy cycle inside the batch loop, though still running in a dedicated scanning thread.  If that doesn't work, I will go back to the "fresh thread every time" approach. I'll also turn on full logging.

I'll let you know if we get success.