dynarithmic/twain_library #164: When to issue DTWAIN_FeedPage inside a validation loop?

dmccallie, 2025-06-28:
I'm using (python) callbacks and capturing `DTWAIN_TN_QUERYPAGEDISCARD` events, fetching the DIB, validating the DIB, saving it as a JPEG, and then if the validation was successful, I return `0` to tell DTWAIN to not save the DIB. 

When a scan fails to validate, I set a `stop` signal and return `0` as above.  When the next `DTWAIN_TN_PAGECONTINUE` callback is received, the code sees the `stop` signal and returns `2` to stop the feeder.  This works, but the scanner will have already processed a few pages before the feeder stops. The operator has to find those fed-but-not-yet-scanned pages, restack them in the hopper, and restart the scanning.

This is OK, but in an attempt to reduce operator work, we want to experiment with "manual" feeding of pages, so that the next page won't get fed for scanning until the previous page has been validated.  This would obviously dramatically slow down the throughput, but we think it might be worth it to reduce operator errors  (this is a ballot-counting system, so errors can't be tolerated.)

My question (finally!) - When should I issue the `DTWAIN_FeedPage()` command to indicate I am ready to continue scanning, assuming that I have already issued:
```
DTWAIN_EnableAutoScan(source, false)
DTWAIN_EnableAutoFeed(source, false)
```

The documentation suggests it should be issued after `DTWAIN_TN_TRANSFERDONE `, but my trace of the callbacks suggests that if I have responded to a `DTWAIN_TN_QUERYPAGEDISCARD` event, the next meaningful event woudl be `DTWAIN_TN_PAGECONTINUE`.  Could I issue the FeedPage call in response to that event?

Here's a snippet of calls coming into my callback:
```
Scanner: DTWAIN_TN_PROCESSEDDIB -> Unprocessed callback event received.
Scanner: DTWAIN_TN_PROCESSEDDIBFINAL -> Unprocessed callback event received.
Scanner: DTWAIN_TN_QUERYPAGEDISCARD -> Page discard? query received.
Image Saved: image_000002.jpg
Scanner: DTWAIN_TN_PAGEDISCARDED -> Unprocessed callback event received.
Scanner: DTWAIN_TN_PAGECONTINUE -> Page continue signal received.
Scanner: DTWAIN_TN_TRANSFERREADY -> Page transfer ready signal received.
Scanner: DTWAIN_TN_TRANSFERDONE -> Unprocessed callback event received.
Scanner: DTWAIN_TN_PROCESSEDDIB -> Unprocessed callback event received.
Scanner: DTWAIN_TN_PROCESSEDDIBFINAL -> Unprocessed callback event received.
Scanner: DTWAIN_TN_QUERYPAGEDISCARD -> Page discard? query received.
Image Saved: image_000003.jpg
```
Sorry if this doesn't make sense!

Thanks.

dmccallie, 2025-06-28:
OK, good.  The printer is remote so it's difficult for me to experiment :(

Assuming that I have set `DTWAIN_EnableAutoScan` to False, and `DTWAIN_EnableAutoFeed` to False, and have been able to set `DTWAIN_SetMaxBuffers` to 1, and that `CAP_FeedPage` is supported, then:

1. how do I get the first page to feed (before the callback is ever called)? Should I call it on `DTWAIN_TN_ACQUIRESTARTED` or will the first FeedPage be called by DTWAIN outside the callback to kick off the acquisition?
2. where in the callback should I call FeedPage to get the next page to be sent?

The documentation suggests calling it after  `DTWAIN_TN_TRANSFERDONE`, so that's what I will try first.

Thanks again.

dynarithmic, 2025-06-29:
I didn't mention this previously, but you should make sure that the capability CAP_FEEDPAGE, CAP_CLEARPAGE, etc. are negotiable when the Source is acquiring images.  You can determine this by calling [DTWAIN_EnumExtendedCaps](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_EnumExtendedCaps.html) to see if those values appear in the DTWAIN_ARRAY.  If they do not appear, then the feeding of the pages is probably not supported (you can still try feeding pages, but officially the Source has to report that these capabilities can be set while acquiring images).

> how do I get the first page to feed (before the callback is ever called)? Should I call it on DTWAIN_TN_ACQUIRESTARTED or will the first FeedPage be called by DTWAIN outside the callback to kick off the acquisition?

When the DTWAIN_TN_ACQUIRESTARTED notification is sent, the Twain message loop has not started (yet).  So sending a "feed page" at this point would be the only viable place to test this.  I would hazard a guess that when the Source gets a "feed page", all it really does is put an event onto the Twain message queue that will be eventually picked up by DTWAIN, and thus start the actual transfer.

Then for the next page, DTWAIN_TN_TRANSFERDONE would be where it probably should be placed, given that (hopefully) all that the Source will do is post a pending message on the message queue that will be eventually picked up in the message processing, starting off the next transfer, etc.

dmccallie, 2025-06-30:
We can't get the one page at a time stuff to work, so we are just going to go with standard autoscan pipeline.

When we detect a validation failure, we signal the feeder to stop (from the callback), and it usually stops within a page or two.
That's good enough for the operators.  We hope.

Thanks!

closed 2025-06-30 by dmccallie