dynarithmic/twain_library #167: imprinter - TWAIN Condition Code: TWAIN triplet is out of expected sequence

dmccallie, 2025-07-09:
Hello again,

Trying to get an imprinter to work on an older Fujitsu 5900C scanner. Using a very old SANE driver, this scanner's imprinter works just fine, so I know it's supported by the hardware.

Here are the `*_PRINTER` capabilities listed in the JSON output of the demo32 program:
```
      "imprinter-supported": false,
 ..... 
          "name": "CAP_PRINTER",
          "value": 4134,
          "type": "standard"
        },
        {
          "name": "CAP_PRINTERENABLED",
          "value": 4135,
          "type": "standard"
        },
        {
          "name": "CAP_PRINTERINDEX",
          "value": 4136,
          "type": "standard"
        },
        {
          "name": "CAP_PRINTERMODE",
          "value": 4137,
          "type": "standard"
        },
        {
          "name": "CAP_PRINTERSTRING",
          "value": 4138,
          "type": "standard"
        },
        {
          "name": "CAP_PRINTERSUFFIX",
          "value": 4139,
          "type": "standard"
        },
```
In my python code, I was able to successfully call `DTWAIN_EnumTwainPrinters`, `DTWAIN_ArrayGetCount` (returns 1), and then `DTWAIN_ArrayGetAtLong` (returns 0 for the 0-the element)  There is only one imprinter so I assume '0' makes sense?

After that, for some reason `DTWAIN_SetPrinter(source, 0, True)` as well as `DTWAIN_SetPrinter(source, 0, False)` both fail (call returns False) with last error message = 0

However, a subsequent `DTWAIN_GetPrinter` successfully returns 0 as the printer number
then `DTWAIN_EnablePrinter` returns successfully.

Then I try to call `DTWAIN_SetPrinterStartNumber` during the callbacks, passing in the "image number" that should be imprinted on each scan.  I call this the first time in response to `DTWAIN_TN_ACQUIRESTARTED` and then subsequently in response to `DTWAIN_TN_PAGECONTINUE`. 

In both these cases `DTWAIN_SetPrinterStartNumber()` fails with the error `Code -1109 - TWAIN Condition Code: TWAIN triplet is out of expected sequence`

Needless to say, nothing prints (though scanning works fine.)

Suggestions?

dmccallie, 2025-07-10:
Thanks.

I realize you can't test or inspect driver code, but do you think that setting a printer _string_ to would work, based on the CAP_PRINTERSTRING capacity being listed?  We'd actually like to format the image number into a standard string with a specific format, so that's actually a better solution than just setting the printer index. Something like "00n-00m"

Does calling these imprinter messages from within the Callback make sense? Will that be the right Twain "state?"

Does that error message "Code -1109 - TWAIN Condition Code: TWAIN triplet is out of expected sequence" give us any clue?

dynarithmic, 2025-07-10:
> Does calling these imprinter messages from within the Callback make sense? Will that be the right Twain "state?" 

Try setting the capabilities directly after opening the Source.  The `DTWAIN_ACQUIRESTARTED` is sent during state 5 (source is ready to acquire), and capability negotiation is limited in states 5, 6, and 7 (only the caps that are extended caps are supported for states 5/6/7 negotiation.

dmccallie, 2025-07-10:
We'd like to be able to set the Printer String for each image dynamically, per image. We'd know the right string right before the scan begins.  Is that likely possible?  When using callbacks, when is the device in state 4?

dynarithmic, 2025-07-10:
> We'd know the right string right before the scan begins. Is that likely possible? 

Unless the CAP_PRINTER... capabilities are also extended capabilities (if the Source supports extended capabilities), negotiating these caps during states 5, 6, and 7 is not possible.  

Going by the JSON listing, the CAP_PRINTER... capabilities are listed as only "standard", so they shouldn't be able to be negotiated while in states 5 - 7.  So you have to negotiate these caps when the UI is closed, or if no UI, before the scan has started and after all pages have been scanned.  

So in general try setting those caps, 

1) before the DTWAIN_Acquire... call or after the call has returned, or 
2) one of the callbacks listed in state 4 (below).

> When using callbacks, when is the device in state 4?

`DTWAIN_TN_ACQUIREDONE`, `DTWAIN_TN_ACQUIRECANCELED`, `DTWAIN_TN_UIOPENING` and `DTWAIN_TN_UICLOSED` are the ones that should be sent when the source is in state 4.